princeton-vl/DPVO #1: how to make calib file for  our own camera

Jiakui, 2022-08-10:
Dear author,

how to make calib file for  our own camera  ?  Can you interprete each number of the calib file so that we can make our  own calib file ?

Thanks

karnikram, 2022-08-10:
According to this [function](https://github.com/princeton-vl/DPVO/blob/41efc1ef77862685f4c0acc6079f655445220de0/dpvo/stream.py#L6), the first four values are the camera intrinsics (fx, fy, cx, cy). The other four values, if provided, are the distortion coefficients (k1, k2, p1, p2).
You can use any standard camera calibration toolbox to obtain these parameters for your camera.

zachteed, 2022-08-11:
Yes that's correct, intrinsics followed by OpenCV camera model. I'll add some documentation of the calibration format to the readme.

closed 2022-08-11 by zachteed