princeton-vl/DPVO #4: bad result on kitti

whitesockcat, 2022-08-12:
[image]
[image]

is there any way to solve this scale problem?

zachteed, 2022-08-12:
Yeah that doesn’t look good, thanks for bringing this up. It looks like there is some bias in the driving scenarios causing the scale to constantly expand.

I will look into this and reply to this thread in a few days. There might be a simple fix, maybe not. The hacky solution would be to impose a prior on depth of the road immediately in front of the car but that’s not ideal

whitesockcat, 2022-08-12:
Thx😀, looking forward to getting your update soon

aniket-gupta1, 2022-10-14:
@whitesockcat what stride did you use for running on kitti?

whitesockcat, 2022-10-20:
> @whitesockcat what stride did you use for running on kitti?
1

RaymonHE, 2022-11-26:
> [image] [image]
> 
> is there any way to solve this scale problem?

Hi @whitesockcat , thanks for sharing the result on KITTI. I am new to KITTI and neural network. I wonder how you test this algorithm on KITTI datasets. Do you just use KITTI as the input and implement the demo.py as shown in the README file, the same way as TartanAir, like `python demo.py --imagedir=<path to image_left> --calib=calib/tartan.txt --stride=1 --viz `, what's the format and how you set the calib and stride file? I would be very grateful if you can give me some suggestion.

nnop, 2022-12-31:
Any fix on this issue?