princeton-vl/DPVO #6: How to add our own training data?

Hongdou68, 2022-08-31:
Dear author:
        thanks for your wonderful work. when I want to add my own data to train dpvo, however I don't known how to generate .pickle file for my own dataset?  （for example TartanAir.pickle file for tartanair dataset）

dicarne, 2022-09-09:
https://github.com/princeton-vl/DPVO/blob/main/dpvo/data_readers/tartan.py line 68, the `_build_dataset()` can generate `scene_info`, save `scene_info` as `YourScene.pickle`.

A error will occur in https://github.com/princeton-vl/DPVO/blob/main/dpvo/data_readers/rgbd_utils.py line 123, no definition found for this function. And you can find this function in https://github.com/princeton-vl/DROID-SLAM/blob/main/droid_slam/geom/projective_ops.py, copy that to your project.

Other than that, there are just a few issues that need to be fixed, for example, modify the path of the imported module.

I've successfully trained on other datasets!

closed 2022-09-22 by Hongdou68

JingruiYu, 2025-02-01:
Hello, have you successfully fine-tuned DPVO?
I have tried fine-tuning it, but the [results](https://github.com/princeton-vl/DPVO/issues/91) are quite strange.
Would you be willing to give me some guidance?