princeton-vl/DPVO #9: Segmentation fault when run demo and some fix

dicarne, 2022-09-08:
OS: `ubuntu 20.04LTS`
Gcc and g++: `10.3.0`
Conda environment: same as `environment.yml`
Cudnn: `8.3.2`
Driver Version: `510.47.03`
CUDA Version: `11.6`

After installing the dependencies and downloading the model according to the README, run the README
demo command, and it run failed. Without and error message.
```
$ CUDA_LAUNCH_BLOCKING=1 python demo.py --imagedir=movies/IMG_0494.MOV --calib=calib/iphone.txt --stride=5 --viz
Running with config...
BUFFER_SIZE: 2048
GRADIENT_BIAS: False
KEYFRAME_INDEX: 4
KEYFRAME_THRESH: 15.0
MIXED_PRECISION: True
MOTION_DAMPING: 0.5
MOTION_MODEL: DAMPED_LINEAR
OPTIMIZATION_WINDOW: 10
PATCHES_PER_FRAME: 96
PATCH_LIFETIME: 13
REMOVAL_WINDOW: 22
[1]    310720 segmentation fault (core dumped)  CUDA_LAUNCH_BLOCKING=1 python demo.py --imagedir=movies/IMG_0494.MOV   --viz
```
I found some method to fix it:
in `DPViewer/dpviewer/viewer.cpp`
[image]
and
[image]

Then I can run without error(sometimes). 
And I found the following change will be helpful, before the change, there was a certain probability of segmentation fault, after the change, no more segmentation fault. Presumably this is due to pointer initialization.
[image]

Since I commented out the code to calculate the `transformMatrix`, the camera will not be updated.
And I found that whenever I called the calculated code in the loop of this thread, it would definitely error out. Therefore, I create a member function specifically for calculating and updating the `transformMatrix` in main thread.

1. New update `transformMatrix` function (Don't forget to write the function definition and mutex definition):
[image]

2. Lock when render thread draw points and poses:
[image]

3. Call it in python:
[image]

Now, in the visualization view, the camera can render the pose normally.

But I observed that point cloud data was used, but my visualization view only had the camera pose and video, not any points.

I'm not very familiar with gl programming, so just a simple replace of the render point function.
[image]

Finally, I can now run the visualization interface properly on my computer and update the poses and point cloud dynamically!

Strangely, the final point cloud results show a trigonometric appearance. According to the observation, the color and shape of the points near the camera path are in line with the image captured in the video. So what are the two extra channels?
[image]

---
Some unanswered questions: Why, even with the same environment configuration as the author claims, I was unable to run the code of this project until I modified some of the code? I changed several computers and operating systems, and tried in docker, but all failed. I'm curious what the environment configuration is like to run it directly.

lahavlipson, 2022-09-11:
We just created an official docker for DPVO with instructions on how to set it up: https://github.com/princeton-vl/DPVO_Docker

Hopefully this will be much easier; if you run into problems with the docker please follow up and I will try my best to help.

oscarfossey, 2022-10-20:
Hello , thanks for the great repo! I have the same issue when using the docker file.  Although i'm not sure it is the same causes because it refers to some XDG_RUNTIME_DIR.

```
(dpvo) root@laptop:/DPVO# python demo.py --imagedir=movies/IMG_0494.MOV --calib=calib/iphone.txt --stride=5 --viz 
Running with config...
BUFFER_SIZE: 2048
GRADIENT_BIAS: False
KEYFRAME_INDEX: 4
KEYFRAME_THRESH: 15.0
MIXED_PRECISION: True
MOTION_DAMPING: 0.5
MOTION_MODEL: DAMPED_LINEAR
OPTIMIZATION_WINDOW: 10
PATCHES_PER_FRAME: 96
PATCH_LIFETIME: 13
REMOVAL_WINDOW: 22
error: XDG_RUNTIME_DIR not set in the environment.
Segmentation fault (core dumped)
(dpvo) root@laptop:/DPVO# 
```

Launch-on-Titania, 2023-10-03:
> Hello , thanks for the great repo! I have the same issue when using the docker file. Although i'm not sure it is the same causes because it refers to some XDG_RUNTIME_DIR.
> 
> ```
> (dpvo) root@laptop:/DPVO# python demo.py --imagedir=movies/IMG_0494.MOV --calib=calib/iphone.txt --stride=5 --viz 
> Running with config...
> BUFFER_SIZE: 2048
> GRADIENT_BIAS: False
> KEYFRAME_INDEX: 4
> KEYFRAME_THRESH: 15.0
> MIXED_PRECISION: True
> MOTION_DAMPING: 0.5
> MOTION_MODEL: DAMPED_LINEAR
> OPTIMIZATION_WINDOW: 10
> PATCHES_PER_FRAME: 96
> PATCH_LIFETIME: 13
> REMOVAL_WINDOW: 22
> error: XDG_RUNTIME_DIR not set in the environment.
> Segmentation fault (core dumped)
> (dpvo) root@laptop:/DPVO# 
> ```
> 
> EDIT: I change in the docker file the the repo link to the one from @dicarne and it worked.
> 
> https://github.com/dicarne/DPVO
> 
> Thank you very much both of you!

Hi, I face the same issue while I use the docker env from the author. Have you solved it?

caoxudong0513, 2024-12-26:
> OS: `ubuntu 20.04LTS` Gcc and g++: `10.3.0` Conda environment: same as `environment.yml` Cudnn: `8.3.2` Driver Version: `510.47.03` CUDA Version: `11.6`
> 
> After installing the dependencies and downloading the model according to the README, run the README demo command, and it run failed. Without and error message.
> 
> ```
> $ CUDA_LAUNCH_BLOCKING=1 python demo.py --imagedir=movies/IMG_0494.MOV --calib=calib/iphone.txt --stride=5 --viz
> Running with config...
> BUFFER_SIZE: 2048
> GRADIENT_BIAS: False
> KEYFRAME_INDEX: 4
> KEYFRAME_THRESH: 15.0
> MIXED_PRECISION: True
> MOTION_DAMPING: 0.5
> MOTION_MODEL: DAMPED_LINEAR
> OPTIMIZATION_WINDOW: 10
> PATCHES_PER_FRAME: 96
> PATCH_LIFETIME: 13
> REMOVAL_WINDOW: 22
> [1]    310720 segmentation fault (core dumped)  CUDA_LAUNCH_BLOCKING=1 python demo.py --imagedir=movies/IMG_0494.MOV   --viz
> ```
> 
> I found some method to fix it: in `DPViewer/dpviewer/viewer.cpp` [image] and [image]
> 
> Then I can run without error(sometimes). And I found the following change will be helpful, before the change, there was a certain probability of segmentation fault, after the change, no more segmentation fault. Presumably this is due to pointer initialization. [image]
> 
> Since I commented out the code to calculate the `transformMatrix`, the camera will not be updated. And I found that whenever I called the calculated code in the loop of this thread, it would definitely error out. Therefore, I create a member function specifically for calculating and updating the `transformMatrix` in main thread.
> 
>     1. New update `transformMatrix` function (Don't forget to write the function definition and mutex definition):
>        [image]
> 
>     2. Lock when render thread draw points and poses:
>        [image]
> 
>     3. Call it in python:
>        [image]
> 
> 
> Now, in the visualization view, the camera can render the pose normally.
> 
> But I observed that point cloud data was used, but my visualization view only had the camera pose and video, not any points.
> 
> I'm not very familiar with gl programming, so just a simple replace of the render point function. [image]
> 
> Finally, I can now run the visualization interface properly on my computer and update the poses and point cloud dynamically!
> 
> Here is all changes: [dicarne@5f17684](https://github.com/dicarne/DPVO/commit/5f176844ecaf75a335c07ff667876de7907f3373) [dicarne@677b056](https://github.com/dicarne/DPVO/commit/677b0567d1b22470f6dcd7a63e03abdd49f3ab29)
> 
> Some unanswered questions: Why, even with the same environment configuration as the author claims, I was unable to run the code of this project until I modified some of the code? I changed several computers and operating systems, and tried in docker, but all failed. I'm curious what the environment configuration is like to run it directly.
can you install dpvo in your environment? what can i do ?  
cuda 11.6
name: dpvo
channels:
  - pyg
  - pytorch
  - nvidia
  - conda-forge
  - defaults
dependencies:
  - pip
  - python=3.8
  - pytorch=2.3.1
  - pytorch-scatter=2.1.2
  - pytorch-cuda=12.1
  - torchvision=0.18
  - pip:
    - tensorboard
    - numba
    - tqdm
    - einops
    - pypose
[... 83 lines cut ...]
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/setuptools/_distutils/cmd.py", line 316, in run_command
          self.distribution.run_command(command)
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/setuptools/dist.py", line 999, in run_command
          super().run_command(command)
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/setuptools/_distutils/dist.py", line 973, in run_command
          cmd_obj.run()
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 98, in run
          _build_ext.run(self)
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 359, in run
          self.build_extensions()
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 522, in build_extensions
          _check_cuda_version(compiler_name, compiler_version)
        File "/root/miniconda3/envs/dpvo/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 417, in _check_cuda_version
          raise RuntimeError(CUDA_MISMATCH_MESSAGE.format(cuda_str_version, torch.version.cuda))
      RuntimeError:
      The detected CUDA version (11.6) mismatches the version that was used to compile
      PyTorch (12.1). Please make sure to use the same CUDA versions.
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for dpvo
  Running setup.py clean for dpvo
Failed to build dpvo
`

SakuraLiHe, 2024-12-27:
@dicarne I have changed the file, how can I rebuild it?

dicarne, 2024-12-30:
> @dicarne I have changed the file, how can I rebuild it? Traceback (most recent call last): File "/DPVO/demo.py", line 87, in (poses, tstamps), (points, colors, calib) = run(cfg, args.network, args.imagedir, args.calib, args.stride, args.skip, args.viz, args.timeit) File "/root/miniconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context return func(*args, **kwargs) File "/DPVO/demo.py", line 50, in run slam(t, image, intrinsics) File "/DPVO/dpvo/dpvo.py", line 388, in **call** self.viewer.loop() AttributeError: 'dpviewerx.Viewer' object has no attribute 'loop'

It's been a while since I remember clearly, but I should have executed `pip install .` in the Viewer directory.