princeton-vl/DPVO #10: Some problems when running demo.py

heng94, 2022-10-06:
Hi, Thanks for opening source your code.

I follow your instructions to configure the environment successfully.
But when running the demo.py, it gives me some errors like this:

_Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.Aborted (core dumped)_

I have searched for it on the internet, but their solutions are not working. 
Can you give me some advice?

Thanks.

lahavlipson, 2022-10-06:
Have you tried running the demo visualization in [our docker container](https://github.com/princeton-vl/DPVO_Docker)?

On some machines I tested, the DPVO visualization would throw this error, but it still ran smoothly in the docker container. I'm not sure what's the root cause, though.

heng94, 2022-10-07:
Because I created the environment successfully by using _conda_, I did not try the docker.
I am going to try to use your docker container.

Thanks.

HonglinChu, 2023-12-23:
> Because I created the environment successfully by using _conda_, I did not try the docker. I am going to try to use your docker container.
> 
> Thanks.

I have same problem, and have you solved it ?

jywu511, 2024-02-22:
The same problem

jialongLiu, 2024-02-27:
I also encountered the same problem, and this problem also exists in docker. Have you solved it?

jywu511, 2024-02-28:
@jialongLiu I found that If I use Linux PC, it is OK to running demo.py, however, when I use SSH connection, It fails. I think it is the problem of Pangolin.