princeton-vl/DPVO #14: Error when implementing demo

RaymonHE, 2022-11-20:
Hi, @zachteed @lahavlipson, sorry to bother you again. After almost one-week-long, I made some breakthroughs. After I created the virtual environment on my Ubunutu20.04 and succeeded in installing CUDA 11.8 and the corresponding cudnn (8.50 for CUDA 11.X), I finally solved the problem of the failure to download DPViewer, which is not that easy as it sounded, especially for me who might had no background of how to implement learning on Ubuntu before. I still encountered some problems when testing the demo given. After I inputted the `python demo.py --imagedir=movies/IMG_0494.MOV --calib=calib/iphone.txt --stride=5 --viz`, the outcome is given below. The error is the failure to locate the file "dpvo.pth", I wonder if I miss something or not. I would be very grateful if you could do me a favor.
`(dpvo) ➜  DPVO git:(main) ✗ python demo.py --imagedir=movies/IMG_0494.MOV --calib=calib/iphone.txt --stride=5 --viz 
Running with config...
BUFFER_SIZE: 2048
GRADIENT_BIAS: False
KEYFRAME_INDEX: 4
KEYFRAME_THRESH: 15.0
MIXED_PRECISION: True
MOTION_DAMPING: 0.5
MOTION_MODEL: DAMPED_LINEAR
OPTIMIZATION_WINDOW: 10
PATCHES_PER_FRAME: 96
PATCH_LIFETIME: 13
REMOVAL_WINDOW: 22
Traceback (most recent call last):
  File "/home/rui/workspace/DPVO/demo.py", line 78, in <module>
    run(cfg, args.network, args.imagedir, args.calib, args.stride, args.skip, args.viz, args.timeit)
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/home/rui/workspace/DPVO/demo.py", line 42, in run
    slam = DPVO(cfg, network, ht=image.shape[1], wd=image.shape[2], viz=viz)
  File "/home/rui/workspace/DPVO/dpvo/dpvo.py", line 21, in __init__
    self.load_weights(network)
  File "/home/rui/workspace/DPVO/dpvo/dpvo.py", line 93, in load_weights
    state_dict = torch.load(network)
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/serialization.py", line 699, in load
    with _open_file_like(f, 'rb') as opened_file:
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/serialization.py", line 230, in _open_file_like
    return _open_file(name_or_buffer, mode)
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/serialization.py", line 211, in __init__
    super(_open_file, self).__init__(open(name, mode))
FileNotFoundError: [Errno 2] No such file or directory: 'dpvo.pth'

`

RaymonHE, 2022-11-20:
I globally searched for the "dpvo.pth", and it showed up in demo.py, line62, evaluate_euroc.py and evaluate_tartan.py. But where is this file and how to find it?

a46068902, 2022-11-20:
You can download these model weights by running `download_models_and_data.sh` from the README

lahavlipson, 2022-11-20:
You can download these model weights by running download_models_and_data.sh from the README

RaymonHE, 2022-11-22:
> You can download these model weights by running download_models_and_data.sh from the README

Hi Lahav, thanks for your advice. When I created the environment, I tried `sh ./download_models_and_data.sh`, and the output is the failure to connect to the network, I didn't why since my network connection is stable. So I figured out how to fix it,  i took a look at the download_models_and_data.sh, i found "wget" and succeeded in installing it, but it failed again, so i just download the TartanAir.pickle and then moved it to datasets and downloaded other to zip file and unzip it in the DPVO folder, which is exactly what the "download_models_and_data.sh" accomplish. I thought i have downloaded all of them, but when i run the final step, something went wrong as i mentioned forward initially. I wonder how to address this problem.

RaymonHE, 2022-11-26:
> You can download these model weights by running download_models_and_data.sh from the README

`Running with config...
BUFFER_SIZE: 2048
GRADIENT_BIAS: False
KEYFRAME_INDEX: 4
KEYFRAME_THRESH: 15.0
MIXED_PRECISION: True
MOTION_DAMPING: 0.5
MOTION_MODEL: DAMPED_LINEAR
OPTIMIZATION_WINDOW: 10
PATCHES_PER_FRAME: 96
PATCH_LIFETIME: 13
REMOVAL_WINDOW: 22
Traceback (most recent call last):
  File "/home/rui/workspace/DPVO/demo.py", line 78, in <module>
    run(cfg, args.network, args.imagedir, args.calib, args.stride, args.skip, args.viz, args.timeit)
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/autograd/grad_mode.py", line 27, in decorate_context
    return func(*args, **kwargs)
  File "/home/rui/workspace/DPVO/demo.py", line 42, in run
    slam = DPVO(cfg, network, ht=image.shape[1], wd=image.shape[2], viz=viz)
  File "/home/rui/workspace/DPVO/dpvo/dpvo.py", line 21, in __init__
    self.load_weights(network)
  File "/home/rui/workspace/DPVO/dpvo/dpvo.py", line 93, in load_weights
    state_dict = torch.load(network)
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/serialization.py", line 699, in load
    with _open_file_like(f, 'rb') as opened_file:
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/serialization.py", line 230, in _open_file_like
    return _open_file(name_or_buffer, mode)
  File "/home/rui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/serialization.py", line 211, in __init__
    super(_open_file, self).__init__(open(name, mode))
FileNotFoundError: [Errno 2] No such file or directory: 'dpvo.pth'

`

RaymonHE, 2022-11-26:
Hi, Lahav,  I solved the problem by adding the absolute path of the "demo.py".  I manually downloaded the dataset in the beginning.  But the problem is the path of the "--network", in line 62, demo.py. I changed the path to the absolute path, and the algorithm worked out.
[image]

> You can download these model weights by running download_models_and_data.sh from the README

viciopoli01, 2026-09-14:
Hello @lahavlipson , It looks like the models are not available anymore on the dropbox. How can I get them?

Thanks.