princeton-vl/DPVO #20: How to understand the softagg for frame aggregation?

ZuoJiaxing, 2023-01-18:
Hi, I've a problem understanding the index of edges when you perform the softagg for the frame aggregation:
net = net + self.agg_ij(net, ii*12345 + jj)
https://github.com/princeton-vl/DPVO/blob/3f65b72dee3c9ed36e7bdb6d106d0ade6d5fc118/dpvo/net.py#L88

In the paper, it is said that frame aggregation is performed on the edges sharing the same source and destination frames. How can it be "ii*12345 + jj"? It will be appreciated if you can explain it a little bit.

lahavlipson, 2023-01-20:
The `SoftAgg` block aggregates edges with the same value in the second argument. "ii*12345 + jj" is the same for all edges with the same source and destination frames.

lahavlipson, 2023-04-03:
`ii*12345 + jj` simply maps a pair of integers, i and j, to single unique integer. The value of this number doesn't matter, `torch.hash(ii, jj)` would work just as well if such a function existed. See https://pytorch-scatter.readthedocs.io/en/1.4.0/composite/softmax.html#torch_scatter.composite.scatter_softmax

senecobis, 2023-04-29:
It's clear thanks