princeton-vl/DPVO #23: BUFFER_SIZE not big enough

relh, 2023-03-15:
```
Traceback (most recent call last):                                                                                                                                                                                                                                                                                                                                                  
  File "/home/relh/DPVO/demo.py", line 90, in <module>                                                                                                                                                                                                                                                                                                                              
    pred_traj = run(cfg, args.network, args.imagedir, args.calib, args.stride, args.skip, args.viz, args.timeit, args.save_reconstruction)                                                                                                                                                                                                                                          
  File "/home/relh/anaconda3/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context                                                                                                                                                                                                                                                                
    return func(*args, **kwargs)                                                                                                                                                                                                                                                                                                                                                    
  File "/home/relh/DPVO/demo.py", line 51, in run                                                                                                                                                                                                                                                                                                                                   
    slam(t, image, intrinsics)                                                              
  File "/home/relh/DPVO/dpvo/dpvo.py", line 342, in __call__                                                                                                                              
    self.index_[self.n + 1] = self.n + 1                                                                                                                                                  
IndexError: index 2048 is out of bounds for dimension 0 with size 2048
```

I have an error from BUFFER_SIZE in the cfg not being large enough. Is the right way to address this to increase the buffer size or is it indicative of a different problem?

relh, 2023-03-16:
For reference the paper mentions this:

```
We connect each patch to every frame within distance r
from the frame index the patch was extracted. This means
that when a new patch is added, we add edges between that
patch and the previous r keyframes. When a new frame is
added, we add edges between each patch extracted in the
last r keyframes with the new frame.
```

It seems unclear whether new patches are connected to the last r keyframes, or connected to keyframes within distance r of the current frame. Regardless my bug just seems to be related to having too long of a video and having a number of keyframes that exceeds the buffer size allocated.

closed 2023-03-16 by relh

lahavlipson, 2023-03-19:
> It seems unclear whether new patches are connected to the last r keyframes, or connected to keyframes within distance r of the current frame.

Every incoming frame is treated as a keyframe, so in the case of DPVO these are the same