princeton-vl/DPVO #33: ERROR: Failed building wheel for dpvo

KcLimitless, 2023-06-14:
I keep getting this error when I run: pip install .  how can I resolve this error ?

garybradski, 2023-11-09:
Do a nvcc --version.  Your CUDA needs to be 11.7
If not, then
conda install -c "nvidia/label/cuda-11.7.0" cuda-toolkit
and
conda install -c "nvidia/label/cuda-11.7.0" cuda

But, I have problems with pangolin and the viewer install. The docker doesn't seem to be a "sufficient" environment, but was inheriting some unspecified things from the original host machine.

JesseZZZZZ, 2024-01-31:
> Do a nvcc --version. Your CUDA needs to be 11.7 If not, then conda install -c "nvidia/label/cuda-11.7.0" cuda-toolkit and conda install -c "nvidia/label/cuda-11.7.0" cuda
>
> But, I have problems with [...]

Thanks! Your reply helps me a lot since I am using a server but I do not have sudo. THX!

curious-energy, 2024-06-20:
I use `conda install -c "nvidia/label/cuda-11.7.0" cuda-toolkit`
`conda install -c "nvidia/label/cuda-11.7.0" cuda`
`conda install -c "nvidia/label/cuda-11.8.0" cudnn`
cmd to **Successfully installed dpviewer-0.0.1**

AdnaneMaj, 2024-07-18:
For those how have cuda installed in the venv ( meaning not a System-wide Installation ) I was able to install the DPviewer by manually adding this line : `set(CMAKE_CUDA_COMPILER "/home/<user_name>/anaconda3/envs/dpvo/bin/nvcc")` to the **CMakeLists.txt** file inside the DPviewer folder. Change <user_name> in order to set the appropriate path to the nvcc flle.

I also faced some issues while trying to build **Pangolin**, specially inside the **DPVO/Pangolin/components/pango_video/include/pangolin/video/drivers/ffmpef_comman.h** file and I was able to solve these by simpling commenting out the following lines : 

```
#ifdef FF_API_XVMC
    //TEST_PIX_FMT_RETURN(XVMC_MPEG2_MC);   //line 60
    //TEST_PIX_FMT_RETURN(XVMC_MPEG2_IDCT); //line 61
#endif
```

Seeeeeyo, 2024-09-06:
Same issue here, does anyone know how to solve this ? I'm using DPVO in WHAM

```
building 'cuda_corr' extension
creating build/temp.linux-x86_64-cpython-39
creating build/temp.linux-x86_64-cpython-39/dpvo
creating build/temp.linux-x86_64-cpython-39/dpvo/altcorr
/usr/bin/gcc-9 -Wno-unused-result -Wsign-compare -DNDEBUG -O2 -Wall -fPIC -O2 -isystem /home/selimgilon/miniconda3/envs/pytorch113/include -I/home/selimgilon/miniconda3/envs/pytorch113/include -fPIC -O2 -isystem /home/selimgilon/miniconda3/envs/pytorch113/include -fPIC -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include/torch/csrc/api/include -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include/TH -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include/THC -I/home/selimgilon/miniconda3/envs/pytorch113/include -I/home/selimgilon/miniconda3/envs/pytorch113/include/python3.9 -c dpvo/altcorr/correlation.cpp -o build/temp.linux-x86_64-cpython-39/dpvo/altcorr/correlation.o -O3 -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=cuda_corr -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
      /home/selimgilon/miniconda3/envs/pytorch113/bin/nvcc -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include/torch/csrc/api/include -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include/TH -I/home/selimgilon/miniconda3/envs/pytorch113/lib/python3.9/site-packages/torch/include/THC -I/home/selimgilon/miniconda3/envs/pytorch113/include -I/home/selimgilon/miniconda3/envs/pytorch113/include/python3.9 -c dpvo/altcorr/correlation_kernel.cu -o build/temp.linux-x86_64-cpython-39/dpvo/altcorr/correlation_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options '-fPIC' -O3 -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE=\"_gcc\" -DPYBIND11_STDLIB=\"_libstdcpp\" -DPYBIND11_BUILD_ABI=\"_cxxabi1011\" -DTORCH_EXTENSION_NAME=cuda_corr -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -ccbin /usr/bin/gcc-9 -std=c++14
      /usr/include/linux/types.h:12:27: error: expected initializer before ‘__s128’
         12 | typedef __signed__ __int128 __s128 __attribute__((aligned(16)));
            |                           ^~~~~~
      error: command '/home/selimgilon/miniconda3/envs/pytorch113/bin/nvcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for dpvo
  Running setup.py clean for dpvo
Failed to build dpvo
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (dpvo)

```

deepConnectionism, 2024-09-11:
@Seeeeeyo you can use docker and python==3.10