princeton-vl/DPVO #36: Cannot use dpviewer on Jetson AGX Orin

ForLoveandPeace, 2023-08-16:
Hi, thanks for your great work! 
I run demo.py on Orin with Ubuntu20.04 and met a error on dpviewer. Here is the output:
`
python demo.py     --imagedir=movies/IMG_0492.MOV      --calib=calib/iphone.txt     --plot     --save_reconstruction     --save_trajectory --viz
Running with config...
BUFFER_SIZE: 2048
GRADIENT_BIAS: False
KEYFRAME_INDEX: 4
KEYFRAME_THRESH: 15.0
MIXED_PRECISION: True
MOTION_DAMPING: 0.5
MOTION_MODEL: DAMPED_LINEAR
OPTIMIZATION_WINDOW: 10
PATCHES_PER_FRAME: 96
PATCH_LIFETIME: 13
REMOVAL_WINDOW: 22
Traceback (most recent call last):
  File "demo.py", line 93, in <module>
    pred_traj = run(cfg, args.network, args.imagedir, args.calib, args.stride, args.skip, args.viz, args.timeit, args.save_reconstruction)
  File "/home/mi/miniforge3/envs/hloc/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "demo.py", line 45, in run
    slam = DPVO(cfg, network, ht=image.shape[1], wd=image.shape[2], viz=viz)
  File "/home/mi/dev/DPVO/dpvo/dpvo.py", line 87, in __init__
    self.start_viewer()
  File "/home/mi/dev/DPVO/dpvo/dpvo.py", line 118, in start_viewer
    from dpviewer import Viewer
  File "/home/mi/miniforge3/envs/hloc/lib/python3.8/site-packages/dpviewer/__init__.py", line 1, in <module>
    from dpviewerx import Viewer
ImportError: /home/mi/miniforge3/envs/hloc/lib/python3.8/site-packages/dpviewerx.cpython-38-aarch64-linux-gnu.so: undefined symbol: _ZN8pangolin19CreateWindowAndBindENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEiiRKNS_6ParamsE
`
Could you help me? Thanks!

build-error, 2024-02-13:
@ForLoveandPeace @jiadeng @zachteed @lahavlipson 

Hello All ! DPVO is a great piece of software, I have been trying to run DPVO on Jetson AGX Orin too !!!

But, I am getting stuck at the same error as @ForLoveandPeace, has anyone found any solution to this issue.

build-error, 2024-02-13:
@ForLoveandPeace @jiadeng @zachteed @lahavlipson 
I have solved the issue, I believe that there is some issue with the included Pangolin files so I decided to remove that one and build from the git repo where Pangolin is originally there. After doing so, the demo was working properly.

Commands that I executed to resolve the issue:
```bash
$ rm -rf Pangolin/
$ git clone --recursive https://github.com/stevenlovegrove/Pangolin.git
$ cd Pangolin/
$ ./scripts/install_prerequisites.sh recommended
$ mkdir build && cd build
$ cmake ..
$ cmake --build .
$ sudo make install
$ pip install ./DPViewer
```

VladimirYugay, 2024-09-08:
Same issue here. The propose fixed didn't solve the problem