princeton-vl/DPVO #37: randomly selecting patch centroids vs. 2D keypoint locations

luopengting, 2023-09-11:
The paper mentions that *Randomly selecting patch centroids works better than 2D keypoint locations produced via SIFT [22], ORB [29], Superpoint [9], or pixels with high image gradient*, I have two questions:

1.  Do you use the classical method to select feature points and resize the location to the feature map size as the patch center centroids?
2.  Is the model retrained when the classical keypoint methods are used?

lahavlipson, 2023-09-11:
1. Yes, exactly.
2. No, these ablations were performed with the model weights trained using random patch centroids.

luopengting, 2023-09-13:
> 1. Yes, exactly.
> 2. No, these ablations were performed with the model weights trained using random patch centroids.

In my opinion, maybe it will be fairer to retrain the model. This paper helps a lot, thank you very much! 😄