princeton-vl/DPVO #39: Unable to find cuda_ba in the site packages folder after running pip install .

YuehaoHuang, 2023-11-02:
Before running the pip install . , running demo.py will prompt no module named cuda_ba
However, after running the pip install .  the program will get stuck in the file dpvo/fastba/ba.py
The IDE marked with a red line in the line "import cuda_ba"
but the program will not report an error, it will only get stuck in the line "ix, jx=fastba. highbors (kk, jj)" in the devo/net.py
Could you please tell me how to solve the problem

YuehaoHuang, 2023-11-02:
If I enter "python" "import cuda_BA" in the terminal, the error message of the terminal is as follows
>>> import cuda_ba
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: libc10.so: cannot open shared object file: No such file or directory

YuehaoHuang, 2023-11-02:
Sorry, I knew.
Because cuda_ba is generated by torch, it is necessary to import torch first

Sadegh-Kalami, 2024-01-10:
in colab u might have problem with ur system path so i add this lines at first of setup.py then everythings goes well.
`
import sys 
for i in ['/usr/local/lib/python3.10/site-packages',
 '/content',
 '/env/python',
 '/usr/lib/python310.zip',
 '/usr/lib/python3.10',
 '/usr/lib/python3.10/lib-dynload',
 '',
 '/usr/local/lib/python3.10/dist-packages',
 '/usr/lib/python3/dist-packages',
 '/usr/local/lib/python3.10/dist-packages/IPython/extensions',
 '/root/.ipython']:
  sys.path.append(i)

print(sys.path)
`

The-Michael-Chen, 2024-04-03:
I have this same issue when I'm runnign the google colab and the code locally

Inglam, 2024-05-18:
> I have this same issue when I'm runnign the google colab and the code locally

Same here. Have you resolved the problem yet?

BavanthaU, 2024-05-21:
I had the same issue due to mismatched CUDA Toolkit and PyTorch versions. The easiest fix is to set up the environment with the correct versions. Use the updated environment.yml file attached below to replace the existing one and test DPVO. It should work without any problems.

[drive link](https://drive.google.com/file/d/10jSdIiEbgFYYN1uieeFwmlkhHlDUxAnw/view?usp=drive_link)