princeton-vl/DPVO #40: Missing calib/icl_nuim.txt

karnikram, 2023-11-02:
Hi, it seems like the camera intrinsics file required by the `evaluate_icl_nuim.py` script is missing. When I use values from the dataset paper (481.20 480.0 319.50 239.50) I get higher ATE values than reported.

Could you share the values used in your paper?

lahavlipson, 2023-11-02:
We used 481.20 -480.00 319.50 239.50

Note that the second number is negative. I'll add the missing icl_nuim.txt

karnikram, 2023-11-03:
Thanks for the clarification. I'd also used the -480 but my problem was I downloaded the wrong version of the dataset from their website (TUM RGB-D compatible PNG instead of ICL-NUIM PNG). It works well with the correct version.

closed 2023-11-03 by karnikram

spokV, 2024-12-02:
for some reason the last trajectory, of-tk3,  results with much higher ate. anyone encounter that?