princeton-vl/DPVO #46: Compling in Widnows

carlosedubarreto, 2024-01-14:
Hello, I'm trying to compile DPVO to use on WHAM project. but I'm not having success, the erorr that shows me is

```
      correlation_kernel.obj : error LNK2001: s¡mbolo externo nÆo resolvido "public: long * __cdecl at::TensorBase::data_ptr<long>(void)const " (??$data_ptr@J@TensorBase@at@@QEBAPEAJXZ)
      build\lib.win-amd64-3.9\cuda_corr.cp39-win_amd64.pyd : fatal error LNK1120: 1 externo nÆo resolvidos
      error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX64\\x64\\link.exe' failed with exit code 1120
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for dpvo
  Running setup.py clean for dpvo
Failed to build dpvo
```

Can you guys give some hint on what I should do to make it work?

thanks a lot

ninefivefly, 2024-01-16:
I also encountered this error.

I changed the 'long' in the correlation_kernel.cu, ba_cuda.cu, and ba.cpp codes to 'int', and this error was not reported.

like this
`
const torch::PackedTensorAccessor32<int,1,torch::RestrictPtrTraits> us,
`

I hope the author will give you a suitable solution.

carlosedubarreto, 2024-01-29:
I found an issue where I explain how to fix it 😅
I'm trying to do it on DPVO code.

https://github.com/mit-han-lab/torchsparse/issues/12#issuecomment-1286733835

carlosedubarreto, 2024-01-29:
and it worked, I had to change the files
correlation_kernel.cu
ba.cpp
ba_cuda.cu

closed 2024-01-29 by carlosedubarreto

carlosedubarreto, 2024-01-29:
Hello @psiydown .
this "loops twice and ends" is on wham code?

Because I might have the same problem.

Actually It seems to start processing and it ends without any error.
I was thinking it was a problem on my installation, but I coulndt spend more time with it. I plan to get to it later.

If you are talking about the test using the DPVO own code, I didnt test it.

psiydown, 2024-01-30:
Hello @carlosedubarreto
I tested the demo code for both WHAM and DPVO and found the same issue. Tracked to, second loop, \dpvo\dpvo.py: if self.n>0 and not self.is_ Initialized:  Print out: n=1, is_ Initialized=False, motion_probe()=NULL, The return has ended.
Can we work together to identify the cause of the problem? Thank you.

carlosedubarreto, 2024-01-30:
Oh, interesting @psiydown.

It would be great to work together to find how to make it work, the problem is I'm not sure when I'm able to go further on WHAM/DPVO code, since I have a bunch of late work to finish, but we can at least keep sharing information about it. I'll update the issue with things I find, and if you can do that too, it could be great.

For example, right now for me to get back on testing I woudl have to have a lot of free hours, since I have no idea on what to do. If i have an idea I can go in with few hours free because I have already a plan on what to do. As I don have any idea, I dont even try to do something, or else I'll surely spend lots of hours and have my late work, even more late LOL.

If you like I have a discord server that we could use to exchange information in a more dynamic way, if you like. or if you have other idea, please let me know.

psiydown, 2024-01-30:
Hello, @carlosedubarreto 
I found through testing that, It exited while executing the 'motion_probe()' function, and did not reach the return step. I am not familiar with the purpose of this function, Is there a solution to replace this function?
https://github.com/princeton-vl/DPVO/blob/main/dpvo/dpvo.py#L376
https://github.com/princeton-vl/DPVO/blob/main/dpvo/dpvo.py#L205
If you find any questions or ideas, you can also answer them in [this question](https://github.com/princeton-vl/DPVO/issues/49). Thank you for taking the time.

carlosedubarreto, 2024-01-30:
@psiydown , I was running the google colab version of  WHAM, and I'll reinstall it on my pc, because I did many different things and probably messed up something.

I'll try to debug that part and let you know if I find something. thanks for all the info.

carlosedubarreto, 2024-01-30:
Great I've been debugging and got to the same point that you showed, but I'm testing it inside WHAM (which seems to be the same problem as I think you said.
[image]

carlosedubarreto, 2024-01-30:
Actually it quits when it gets to the reproject part
[image]

carlosedubarreto, 2026-05-09:
Just as an update. Here is the result from the problem, how to make it compile and not crash.
https://github.com/princeton-vl/DPVO/issues/49#issuecomment-1922447151