princeton-vl/DPVO #47: Points not visible in DPViewer

akashsharma02, 2024-01-26:
Hello, 

Thanks for your excellent work, and for making it open source! I am able to build the repository and run the examples, however I notice that the `DPViewer` visualizer does not show any 3D points, which I expect to see, since 3D points are visible in the Youtube videos and other demo videos. 

[image]

lahavlipson, 2024-01-29:
I can't seem to reproduce the issue. This could be an OpenGL problem, assuming the [`points`](https://github.com/princeton-vl/DPVO/blob/58338357611e7cf3563ed90f82168da7f8741d24/DPViewer/dpviewer/viewer.cpp#L116) tensor in viewer.cpp is still being populated with nonzero values. Maybe try `std::cout << points << std::endl;` to confirm.

akashsharma02, 2024-02-07:
I find that all the points are populated with zeros