princeton-vl/DPVO #49: After executing demo.py, while only loops twice and ends

psiydown, 2024-01-29:
The result was that the file was not saved and there were no error messages. What is the reason?
@jiadeng

lahavlipson, 2024-01-29:
The most likely reason is that the input data was not found.  Can you double check that the video_stream or image_stream is actually loading the images?

psiydown, 2024-01-30:
Hello @lahavlipson 
I have tested that the images and videos are correct and inputted, and can print out the shape of the images. What I want to emphasize is that installing DPVO on Windows does not support the long type. I need to change the long type of the following three files to the long long type in order to compile and install them successfully.
correlation_kernel.cu
ba.cpp
ba_cuda.cu

I'm not sure if changing the long long type is causing it to not work properly, or if DPVO is not compatible with Windows.  Can you identify the cause of the problem or test compatibility with Windows? Thank you very much!

lahavlipson, 2024-01-30:
I wasn't familiar with this issue, thanks for letting me know; I will investigate.

carlosedubarreto, 2024-01-30:
I did go further and with me it was exiting on this line (I'm also on windows and had to compile hanging the source from long to int64_t (i think it was this):

https://github.com/princeton-vl/DPVO/blob/58338357611e7cf3563ed90f82168da7f8741d24/dpvo/dpvo.py#L188

it stops running wihtout any error.

psiydown, 2024-01-30:
@carlosedubarreto Yes, you are correct. It actually exited at this line:
https://github.com/princeton-vl/DPVO/blob/58338357611e7cf3563ed90f82168da7f8741d24/dpvo/projective_ops.py#L60
motion_probe()->reproject()->pops.transform()

carlosedubarreto, 2024-01-30:
Even more specific, while debugging, the error happens at exactly this point

`poses[:, ii].inv() `

If i watch this part, the process crashes also while debugging giving no error.

[image]

I'm starting to think it might be something related with pytorch. I plan to test on other pytorch version. Since I was using to process wham, I'm with pytorch 1.11 cuda 11.3

psiydown, 2024-01-30:
@carlosedubarreto 
I am also using WHAM, pytorch 1.11 cuda 11.3 is the same as your version.

I captured the entire process of error occurrence using `python -X faulthandler demo.py`, 

looking from bottom to top:
```
Current thread 0x000044f4 (most recent call first):
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\lietorch\group_ops.py", line 14 in forward
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\lietorch\groups.py", line 129 in apply_op
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\lietorch\groups.py", line 147 in inv
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\projective_ops.py", line 60 in transform
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\dpvo.py", line 188 in reproject
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\dpvo.py", line 212 in motion_probe
  File "N:\Python\WHAM-main\third-party\DPVO\dpvo\dpvo.py", line 376 in __call__
  File "N:\Python\WHAM-main\third-party\DPVO\demo.py", line 51 in run
  File "N:\anaconda3\lib\site-packages\torch\autograd\grad_mode.py", line 27 in decorate_context
  File "N:\Python\WHAM-main\third-party\DPVO\demo.py", line 92 in <module>
```

carlosedubarreto, 2024-01-30:
great finding @psiydown .

so probably the guy that is causing problems is the compilation of lietorch... 
I'll try to compile it using some other option instead of the "int64_t" replacong the "long" word.

psiydown, 2024-01-30:
@carlosedubarreto 
I have tried changing 'long' to 'long long' type instead of ‘int64_t’ type, the result is the same.

carlosedubarreto, 2024-01-31:
@lahavlipson , I made a lot of changes to use the pypose library, but it didnt work too.
Nover thought it would be this hard.

thanks for the help

psiydown, 2024-01-31:
@lahavlipson @carlosedubarreto 
[Regarding this issue](https://github.com/princeton-vl/lietorch/issues/22), lietorch cannot be used on Windows 10.He suggest "managed to fix this on Windows by making all the CUDA kernels in lietorch_gpu.cu accept regular pointers where currently they accept const pointers. " But I don't know how to change accept regular pointers, hope your know how to do so.

carlosedubarreto, 2024-01-31:
Another great finding @psiydown .
I dont know much about compiling, but I'll tkae a look on that possibility. thanks a lot.

carlosedubarreto, 2024-02-01:
@psiydown I was able to make it work.

I changes the files from long to int64_t and also removed the "const" from the lietorch_gpu.cu

Here are the files I changed to be able to compile and not crash.

[fastba.zip](https://github.com/princeton-vl/DPVO/files/14131997/fastba.zip)

[lietorch_gpu.zip](https://github.com/princeton-vl/DPVO/files/14131991/lietorch_gpu.zip)

[correlation_kernel.zip](https://github.com/princeton-vl/DPVO/files/14132000/correlation_kernel.zip)

psiydown, 2024-02-03:
@carlosedubarreto 
This is a great discovery, I have succeeded, thank you very much for your help!

Next, I plan to implement real-time camera WHAM inference. If you are interested, if you encounter any problems during the implementation process, we can communicate via email and exchange opinions. My email: pansi@qq.com

closed 2024-02-03 by psiydown

carlosedubarreto, 2026-05-09:
Funny thing. 2 year later, I come back to try DPVO and was struggling for 2 days to make it work... Spent all my Google gemini credits (now I'll be able to use it again next week), and found this message were I sent files ready to compile 😵‍💫

It makes me so happy and angry too LOL.