princeton-vl/DPVO #50: Regarding the issue of deep initialization

FDF521X, 2024-02-19:
Author, hello. I noticed that DPVO samples on the feature map and initializes the deep values randomly. If I want to use sensor depth values for initialization, how should I proceed? (I tried multiplying the coordinates of the sample points by 4 and retrieving the depth values from the depth map, but found that the read depth values were incorrect.)

lbblz, 2024-10-28:
Hi, have you implemented this, and if so, could you tell me the details

lbblz, 2024-10-28:
> 作者，你好。我注意到DPVO在特征图上采样，并随机初始化深度值。如果我想使用传感器深度值进行初始化，我应该怎么做？（我尝试将样本点的坐标乘以4，并从深度图中检索深度值，但发现读取的深度值不正确。）

Hi, have you implemented this, and if so, could you tell me the details

Anonymous-on, 2025-03-01:
> > 作者，你好。我注意到DPVO在特征图上采样，并随机初始化深度值。如果我想使用传感器深度值进行初始化，我应该怎么做？（我尝试将样本点的坐标乘以4，并从深度中检索深度值，但发现读取的深度值不正确。）
> 
> 你好，你已经实现了这个吗？如果是的话，你能告诉我细节吗
你好想问一下你解决了吗