princeton-vl/DPVO #53: How is the structure of the Tartan datasets arranged during training?

Tianci-Wen, 2024-04-02:
Thank you for open sourcing your excellent work. I encountered some problems during training. For one of the sequences, my path is as follows：
～/datasets/TartanAir/abandonedfactory/Easy/depth_left/abandonedfactory/EasyP000/depth_left/xxxx.npy
~/datasets/TartanAir/abandonedfactory/Easy/image_left/abandonedfactory/Easy/P000/image_left/xxxx.png

How can I rearrange the file structure?
Is there any preprocessing after downloading the data set? 
Thank you very much and look forward to your answer.

lahavlipson, 2024-04-02:
The data should be stored as `DPVO/datasets/TartanAir/abandonedfactory/abandonedfactory/Easy/P001/image_left/xxxx.png`

Sorry for the confusion. I believe this is how the tartanair_tools github repo unpacks it

Tianci-Wen, 2024-04-03:
> The data should be stored as `DPVO/datasets/TartanAir/abandonedfactory/abandonedfactory/Easy/P001/image_left/xxxx.png`
> 
> Sorry for the confusion. I believe this is how the tartanair_tools github repo unpacks it

Thank you very much for your patient reply! I found this in the tartanair_tools github repo, but it still doesn't work when testing driod-slam. I tested the first abandonedfactory on DPVO and it works. I will test and train DPVO after all sequences have been downloaded.

JingruiYu, 2025-02-01:
Hello, have you successfully fine-tuned DPVO?
I have tried fine-tuning it, but the [results](https://github.com/princeton-vl/DPVO/issues/91) are quite strange.
Would you be willing to give me some guidance?