princeton-vl/DPVO #57: Trajectory Saving Error + make_traj

DannyTran123, 2024-06-14:
Hi,

I believe there is an issue with the trajectory saving that comes from the way the trajectory is created. Here (https://github.com/princeton-vl/DPVO/blob/main/dpvo/plot_utils.py#L14) when the trajectory is created from the poses it assumes the quaternion coordinates are in the form of qw, qx, qy, qz but other parts of the code indicate that the tensor is actually stored as qx, qy, qz, qw such as https://github.com/princeton-vl/DPVO/blob/main/dpvo/dpvo.py#L80 and https://github.com/princeton-vl/DPVO/blob/main/dpvo/dpvo.py#L155 (here since I believe SE3 assumes the quaternion coordinates are xyzw indicating the poses is actually x,y,z,qx,qy,qz,qw and not x,y,z,qw,qx,qy,qz. This leaves the trajectory to get create improperly in make_traj in dpvo/plot_utils.py and then get saved incorrectly (https://github.com/princeton-vl/DPVO/blob/main/dpvo/plot_utils.py#L51).

xiaojxkevin, 2024-07-20:
> Hi,
>
> I believe there is an issue with the trajectory saving that comes from the way the trajectory is created. Here (https://github.com/princeton-vl/DPVO/blob/main/dpvo/plot_utils.py#L14) when the [...]

@DannyTran123 Hi, I have also some questions about the trajectory saving part. I have tried this algorithm on one of my own dataset, and it turns out that the first line in the output tum file is _0, 0, 0, 0, 0, 0, 1, 0_, which seems contradicting to both the x,y,z,qx,qy,qz,qw and x,y,z,qw,qx,qy,qz since in normal cases qw should be 1, corresponding to the Identity SE3 pose.
As a result, I think the output format of this repo is in fact **timestamp,x,y,z,qy,qz,qw,qx**. And my reason is also [(https://github.com/princeton-vl/DPVO/blob/main/dpvo/plot_utils.py#L51]((https://github.com/princeton-vl/DPVO/blob/main/dpvo/plot_utils.py#L51), since quat in Scipy is stored in qx,qy,qz,qw order. As a result, by indexing in order _[1,2,3,0]_, sequence qy,qz,qw,qx will be returned.

lahavlipson, 2024-08-10:
Thank you for pointing this out. This issue has been fixed in the latest release.

closed 2024-08-10 by lahavlipson