princeton-vl/DPVO #59: ValueError: all input arrays must have the same shape

doujiarui, 2024-07-13:
I encountered this problem during training and I don't know how to solve it. Maybe it's a data issue?

```
Traceback (most recent call last):
  File "/home/doujiarui/DPVO/train.py", line 173, in <module>
    train(args)
  File "/home/doujiarui/DPVO/train.py", line 77, in train
    for data_blob in train_loader:
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 652, in __next__
datasets/TartanAir/japanesealley/japanesealley/Easy/P004/image_left/000000_left.png
-----
    data = self._next_data()
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1327, in _next_data
    return self._process_data(data)
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1373, in _process_data
    data.reraise()
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/_utils.py", line 461, in reraise
    raise exception
ValueError: Caught ValueError in DataLoader worker process 3.
Original Traceback (most recent call last):
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/_utils/worker.py", line 302, in _worker_loop
    data = fetcher.fetch(index)
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 49, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 49, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/data/dataset.py", line 235, in __getitem__
    return self.datasets[dataset_idx][sample_idx]
  File "/home/doujiarui/DPVO/dpvo/data_readers/base.py", line 161, in __getitem__
    images = np.stack(images).astype(np.float32)
  File "<__array_function__ internals>", line 180, in stack
  File "/home/doujiarui/anaconda3/envs/dpvo/lib/python3.10/site-packages/numpy/core/shape_base.py", line 426, in stack
    raise ValueError('all input arrays must have the same shape')
ValueError: all input arrays must have the same shape
```

doujiarui, 2024-07-15:
I have already resolved it because there are empty files in the dataset.

JingruiYu, 2025-02-01:
Hello, have you successfully fine-tuned DPVO?
I have tried fine-tuning it, but the [results](https://github.com/princeton-vl/DPVO/issues/91) are quite strange.
Would you be willing to give me some guidance?