princeton-vl/DPVO #68: dpvo ---colmap

li1u, 2024-08-10:
Hello, I am amazed by the effect of your article. But I have some questions to ask you.

First, when I use video data, the visualization result is good, but the stored result.ply file is not good.
[image]
[image]

Second, there are still some differences between the cameras.txt and other files in the result folder and the standard colmap data. I don’t know whether your code provides an interface for storing as standard colmap data.

lahavlipson, 2024-08-10:
Unfortunately, I am not able to reproduce this issue. If you could point me to an example which reproduces the problem, that would be very helpful for me.

The `--colmap` flag saves a directory which can be opened using:
`colmap gui` -> select `file, import model` -> select `path/to/dpvo-output` -> select **no** to only visualize reconstruction

When I do this on the demo from the colab notebook, I get:

[image]

FYI: The --colmap output does not contain information about the 2D feature tracks, since I assumed most people just wanted the camera poses and maybe a sparse reconstruction. If this would be helpful for people, I could add it.

li1u, 2024-08-12:
Yes, it really works for me, hope you can add it.

ljjTYJR, 2024-09-30:
@lahavlipson By the way, I think there is a minor error when saving the camera intrinsic parameter here:
In the [COLMAP doc](https://colmap.github.io/format.html#:~:text=%23%20%20%20CAMERA_ID%2C%20MODEL%2C%20WIDTH%2C%20HEIGHT%2C%20PARAMS%5B%5D), when saving the camera parameters should be
```
# Camera list with one line of data per camera:
#   CAMERA_ID, MODEL, WIDTH, HEIGHT, PARAMS[]
# Number of cameras: 3
1 SIMPLE_PINHOLE 3072 2304 2559.81 1536 1152
2 PINHOLE 3072 2304 2560.56 2560.56 1536 1152
3 SIMPLE_RADIAL 3072 2304 2559.69 1536 1152 -0.0218531
```

but in the code, it is saved as 
https://github.com/princeton-vl/DPVO/blob/c0c5a104c9c58663aa9be62c3f125d5b52874f3e/dpvo/plot_utils.py#L56

I think it should be 
```
 (colmap_dir / "cameras.txt").write_text(f"1 PINHOLE {W} {H} {fx} {fy} {cx} {cy}") 
```

lahavlipson, 2024-10-12:
@ljjTYJR Thanks for catching that. It's now fixed

htchr, 2025-04-07:
> Unfortunately, I am not able to reproduce this issue. If you could point me to an example which reproduces the problem, that would be very helpful for me.
> 
> The `--colmap` flag saves a directory which can be opened using: `colmap gui` -> select `file, import model` -> select `path/to/dpvo-output` -> select **no** to only visualize reconstruction
> 
> When I do this on the demo from the colab notebook, I get:
> <img alt="Screenshot 2024-08-10 at 11 18 22 AM" width="1046" src="https://private-user-images.githubusercontent.com/16674837/356831734-f85eae2a-466c-4cca-80eb-21190c76b9f7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDQwMzg1MTcsIm5iZiI6MTc0NDAzODIxNywicGF0aCI6Ii8xNjY3NDgzNy8zNTY4MzE3MzQtZjg1ZWFlMmEtNDY2Yy00Y2NhLTgwZWItMjExOTBjNzZiOWY3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA0MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNDA3VDE1MDMzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThkNmNkNDc4NmU2Y2ZlNzEzNjM5Yzg2MDk2YjllOTEwMzRmYzlkZTgxMjlmYjgxYTMwZTgwNDIzZTZhZWNiN2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.j8TGdV6H3WMhlo1lYMdKTmEAr8XQbFWTR08RTnJxmcQ">
> 
> FYI: The --colmap output does not contain information about the 2D feature tracks, since I assumed most people just wanted the camera poses and maybe a sparse reconstruction. If this would be helpful for people, I could add it.

Hello, thank you for all your contributions. If it's not too much trouble, could you try adding the 2D feature tracks? and/or explaining how I might add them?

yingyexin, 2025-04-25:
For the First Problem:
If anyone suffers from the abnormal reconstruction in *.ply, try to exclude the extreme outlier (e.g. top 10% max norm), you will find it work well :)

jashshah999, 2026-05-10:
For DPVO-to-COLMAP conversion, [gtsam-splatfactors](https://github.com/jashshah999/gtsam-splatfactors) can ingest VO poses and refine them via factor graph optimization with loop closure, then export clean COLMAP format.

https://github.com/jashshah999/gtsam-splatfactors