princeton-vl/DPVO #73: Benchmarking results for DROID-SLAM on KITTI

ThakurSarveshGit, 2024-09-04:
Thanks for sharing the wonderful work. While replicating the results I ecountered an issue with DROID's performance benchmarked on KITTI. It has much higher error in my experiments than in the table 2 in DPVS-SLAM paper, I am not able to figure out why. Since original Droid-SLAM does not shows result on KITTI, I would like to know what metric was used(I kept the one similar to followed in DPV-SLAM paper). However, I got DPV-SLAM performance similar to shown in paper:

[image]

Performance I got for DROID on KITTI:

[image]

Perhaps I am using different parameters, it would be great if you could share you view. Thanks!

lahavlipson, 2024-09-13:
I used the same ATE[m] metric from DROID-SLAM and other prior work, and used the default demo.py settings from droid-slam with stride=3.

ThakurSarveshGit, 2024-09-15:
> I used the same ATE[m] metric from DROID-SLAM and other prior work, and used the default demo.py settings from droid-slam with stride=3.

Thanks for the response. Will update here.