princeton-vl/DPVO #74: Wrong pointcloud color shown in colmap gui

blackjacket996, 2024-09-06:
Hey! I successfully run demo.py --save_colmap and got three files: camera.txt, points3D.txt and images.txt. I wanna show them using colmap gui, and I got a pointcloud with right shape but wrong color.image above is the origin image and the other is the model shown in colmap. Look at the blocks on the ground, there are no purple and orange blocks in the image. And I also try six different combinations of the "RGB" data in the points3D.txt file generated by the demo.py, but none of the combinations  is correct.
[image]
[image]

blackjacket996, 2024-09-12:
I've solved the problem by rewrite the line 50-52 of plot_utils.py:
```
for i, ((x, y, z),(r,g,b) in enumrate(zip((points*scale).tolist(), colors)):
       points3D += f"{x} {y} {z} {r} {g} {b}` + "0.0 0 0 0 0 0 0\n"
```

closed 2024-09-12 by blackjacket996

lahavlipson, 2024-10-12:
Thanks for raising this issue. It seems that your code change skips the "POINT3D_ID" for each line (see [doc](https://colmap.github.io/format.html#points3d-txt)). Without it, colmap seems to crash for me with `Check failed: points3D_.emplace(point3D_id, std::move(point3D)).second`. For reference, I am using COLMAP 3.10-dev

blackjacket996, 2024-10-18:
Sry, I miss the param "i“, it should be:`for i, ((x, y, z),(r,g,b) in enumrate(zip((points*scale).tolist(), colors)):
       points3D += f"{i} {x} {y} {z} {r} {g} {b}` + "0.0 0 0 0 0 0 0\n"`