princeton-vl/DPVO #77: Failed to run evaluate_kitti.py

Tavish-777, 2024-09-24:
[image]
When I run this script, it gets stuck at this point and stops responding. After debugging, I found that it gets stuck at the queue.get() step, but I don't have any clues on how to solve this issue. I hope the author can provide some assistance.

lahavlipson, 2024-09-25:
The dataloading process probably crashed because one of the expected files is not present. By default, the code expects the first sequence to be:
- images: `datasets/KITTI/dataset/sequences/00/image_2/*.png`
- poses: `datasets/KITTI/poses/00.txt`
- intrinsics: `datasets/KITTI/dataset/sequences/00/calib.txt`

which should be what you get if you unpack the [kitti odometry files ](https://www.cvlibs.net/datasets/kitti/eval_odometry.php)under `datasets/KITTI`

Tavish-777, 2024-09-25:
Thanks, now my dataloading is fine, able to read data , but I have a new problem;
[image]
It will interrupt unexpectedly

Tavish-777, 2024-09-25:
I tried deleting and reinstalling the environment, but it seems that the situation has gotten worse.
[image]
It seems that there is a problem with the C++ part, but I don't know how to debug it.

closed 2024-10-09 by Tavish-777