princeton-vl/DPVO #81: There is a problem running evalutae_kitti.py

Tavish-777, 2024-10-09:
[image]
The check found that there was a problem with the value of the self_positions_xyz:
`self._positions_xyz is :
[[-0. -0. -0.]]`
How can I fix this bug, check my data read, ground-truth is fine.

lahavlipson, 2024-10-12:
It seems like only one keyframe was created, meaning the model failed to initialize. This might occur if somehow the data was being read incorrectly.

If you comment-out the model inference and run with `--show_img`, do the images look correct?

Tavish-777, 2024-10-12:
Thank you for your response. I solved the issue by using the version of the code you provided two months ago.

lahavlipson, 2024-10-12:
FYI some of the cuda kernels were updated recently, so you may need to run
`pip uninstall dpvo && pip install .` to get the latest