princeton-vl/DPVO #84: Warning BA failed

Davidyao99, 2024-10-30:
The bundle adjustment is failing and throwing the following exception:

[image]

Anyone have similar experience? I did not change any of the internals and dpvo setup built successfully. Any help or insight is much appreciated. Thank you!

lahavlipson, 2024-10-31:
From the function signature, it looks like the dpvo binaries were installed from a previous version of the code. 

Try: `pip uninstall dpvo && pip install .`

Davidyao99, 2024-10-31:
I did that but I seem to be facing the same error. Is there a way for me to determine if I have installed the latest dpvo?

lahavlipson, 2024-10-31:
If the latest c++/cuda kernels are installed, the function signature you see when running `>>> import torch; import cuda_ba; help(cuda_ba.forward)` should be:
```
forward(...) method of builtins.PyCapsule instance
    forward(arg0: torch.Tensor, arg1: torch.Tensor, arg2: torch.Tensor, arg3: torch.Tensor, arg4: torch.Tensor, arg5: torch.Tensor, arg6: torch.Tensor, arg7: torch.Tensor, arg8: torch.Tensor, arg9: int, arg10: int, arg11: int, arg12: int, arg13: bool) -> List[torch.Tensor]
```
which matches the function signature:
https://github.com/princeton-vl/DPVO/blob/859bbbfdac6c6185f345003b3c473901fcd13ace/dpvo/fastba/ba.cpp#L32

Maybe delete the `build/` folder and `.egg` files before reinstalling if you haven't tried that.

Davidyao99, 2024-10-31:
I see I see, thank you so much. I managed to resolve it by recreating a new environment and reinstalling the repo.

closed 2024-10-31 by Davidyao99