princeton-vl/DPVO #88: Usage of Depth Maps

JesseZZZZZ, 2024-12-03:
Hi, dear authors! 
Thank you for your great work, I have a small issue. In your paper, the depth information is part of the patch. But during inference, the depth maps are initialized all zero, how can this be? Is this because depth is only related to optical flow loss, and not related to trajectory loss, so you don't need depth maps during inference? 

Thank you!

lushouyi, 2024-12-03:
The author did not use the ground truth depth information during the training process. Instead, the depth values were set to random numbers.

JesseZZZZZ, 2024-12-03:
@lushouyi Are you sure...? I believe the `disps` is the inverse depth... Only the depth map durence inference is initialized all zeros

lushouyi, 2024-12-03:
Line 206 of the net.py file

lushouyi, 2024-12-05:
You can set everything to 1 instead, I feel that setting everything to 0 might be problematic.