princeton-vl/DPVO #89: ImportError: libpango_windowing.so: cannot open shared object file: No such file or directory

hseysen, 2024-12-03:
I've tried local build first, then the Docker image. Everything goes fine without any errors. When I run `python demo.py --imagedir=movies/IMG_0492.MOV --calib=calib/iphone.txt --stride=5 --plot --viz` I get:

```
Initialized new /root/.evo/settings.json
Running with config...
BACKEND_THRESH: 64.0
BUFFER_SIZE: 4096
CENTROID_SEL_STRAT: RANDOM
CLASSIC_LOOP_CLOSURE: False
GLOBAL_OPT_FREQ: 15
KEYFRAME_INDEX: 4
KEYFRAME_THRESH: 15.0
LOOP_CLOSE_WINDOW_SIZE: 3
LOOP_CLOSURE: False
LOOP_RETR_THRESH: 0.04
MAX_EDGE_AGE: 1000
MIXED_PRECISION: True
MOTION_DAMPING: 0.5
MOTION_MODEL: DAMPED_LINEAR
OPTIMIZATION_WINDOW: 10
PATCHES_PER_FRAME: 96
PATCH_LIFETIME: 13
REMOVAL_WINDOW: 22
Traceback (most recent call last):
  File "/DPVO/demo.py", line 84, in <module>
    (poses, tstamps), (points, colors, calib) = run(cfg, args.network, args.imagedir, args.calib, args.stride, args.skip, args.viz, args.timeit)
  File "/root/miniconda3/envs/dpvo/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/DPVO/demo.py", line 46, in run
    slam = DPVO(cfg, network, ht=H, wd=W, viz=viz)
  File "/DPVO/dpvo/dpvo.py", line 80, in __init__
    self.start_viewer()
  File "/DPVO/dpvo/dpvo.py", line 115, in start_viewer
    from dpviewer import Viewer
  File "/root/miniconda3/envs/dpvo/lib/python3.10/site-packages/dpviewer/__init__.py", line 1, in <module>
    from dpviewerx import Viewer
ImportError: libpango_windowing.so: cannot open shared object file: No such file or directory
```

Which is strange, because I've built and ran `sudo make install` for Pangolin, and `/usr/local/lib/libpango_windowing.so` exists. Where is the issue stemming from?

2019lelexia, 2025-02-11:
sudo ldconfig

atomishcv, 2025-03-20:
@2019lelexia  it's work!! thank you