princeton-vl/DPVO #91: How to fine-tune DPVO on your own dataset and achieve better results？

JingruiYu, 2025-02-01:
Hello, Great work! so I tried it on my own dataset, but the results are quite strange.
I prepared the data in the format of TartanAir. Under each scene folder, there are _depth_left_ folders, _image_left_ folders, and _pose_left.txt_ file. 
The format of the pose_left.txt file is `tz, tx, ty, qz, qx, qy, qw`. I noticed that the code adjusts the columns when loading the pose txt file.
Additionally, the resolution of my images is different from that of TartanAir, so I modified the code to read the intrinsic parameters from a different location.
After preparing the data, I used the code like [here](https://github.com/princeton-vl/DPVO/blob/main/dpvo/data_readers/tartan.py#L68) to generate the pickle file. 
BTW, I removed the operation `poses[:,:3] /= DEPTH_SCALE`, thinking it wouldn't have a significant impact.
Then, I used a train command almost identical to the one in the readme, except that I reduced the --lr to 0.00004 for fine-tuning. I mixed my own data with the TartanAir data for this process.

However, the trajectory results plotted during the fine-tuning process are quite poor. For example, they look like this. 
step=8000
[image]
step=16000

[image]
step=30000

[image]

But before fine-tuning, the results obtained using the downloaded model were like this.

[image]

In other words, the results have become worse. Under the same training steps, the results on the TartanAir data were perfectly normal.

[image]

I'm not sure where I might have gone wrong or what the reason could be. Thank you!

DOICH89, 2025-03-22:
Hello, have you succeeded in fine-tuning? I also want to adjust it but I don’t know how to do it.

JingruiYu, 2025-03-27:
yes, RGB and Lidar are convertered into rgb-d in Tartan format. But it cannot work. After carefully check the code, I think there are some bugs.

lbblz, 2025-04-09:
> yes, RGB and Lidar are convertered into rgb-d in Tartan format. But it cannot work. After carefully check the code, I think there are some bugs.

What bugs did you find

akramsalim, 2025-04-09:
> Hello, have you succeeded in fine-tuning? I also want to adjust it but I don’t know how to do it.

You donot have to make fine tuning. What you need to do is Zero-Shot or evaluation directly with loading the weights of the pretrained model, for that you can update one of the provided examples like  [valuate_kitti.py](https://github.com/princeton-vl/DPVO/blob/main/evaluate_kitti.py) etc. Just read the paper and you will understand it.

akramsalim, 2025-04-09:
> > yes, RGB and Lidar are convertered into rgb-d in Tartan format. But it cannot work. After carefully check the code, I think there are some bugs.
> 
> What bugs did you find

no bug discovered until now. You donot have to make fine tuning. What you need to do is Zero-Shot or evaluation directly with loading the weights of the pretrained model, for that you can update one of the provided examples like  [valuate_kitti.py](https://github.com/princeton-vl/DPVO/blob/main/evaluate_kitti.py) etc. Just read the paper and you will understand it. Try also to adjust the hyperparameters to get very good results

DOICH89, 2025-04-15:
Hello, I would like to ask how you generate the pickle file of your own dataset. If possible, I would like to refer to your code.

lbblz, 2025-04-16:
> Hello, I would like to ask how you generate the pickle file of your own dataset. If possible, I would like to refer to your code.

https://github.com/princeton-vl/DPVO/issues/6

JingruiYu, 2025-04-16:
> Hello, I would like to ask how you generate the pickle file of your own dataset. If possible, I would like to refer to your code.
I haven't had time to do it yet, will update when I finish.

JingruiYu, 2025-04-16:
> Since Tartanair is not consistent with the common camera coordinate system, I guess this part is a conversion of the camera coordinate system, is there a problem here?
Yes, I also noticed the issue you mentioned. The author's original intention was to swap the coordinate axes. While this operation works in translation, but not in quaternion. It should be performed by left-multiplying the following rotation matrix, 
```
R = [0 0 1
       1 0 0
       0 1 0].
```
These two are not equivalent.
My understanding, open to discussion.

JingruiYu, 2025-10-15:
> [@JingruiYu](https://github.com/JingruiYu) hello，do you finetune successed，i meet the same problem, thankyou!

Hi @jumesnail , I haven't had a chance to test it out yet. I'll let you know once I have some results.